sinara-hw/Stabilizer_Piezo_Driver #2: ADHV4702

dtcallcock, 2019-09-07:
I may need to get something working before this board is ready so I'm planning on ordering an [eval board](https://www.analog.com/media/en/technical-documentation/user-guides/EVAL-ADHV4702-1CPZ-UG-1444.pdf) for the new [ADHV4702](https://www.analog.com/media/en/technical-documentation/data-sheets/ADHV4702-1.pdf).

If that works out I'll let you know.  Perhaps we could just copy this design (and make it 2-ch) if it does.

gkasprow, 2019-12-01:
@dtcallcock any opinion here?
I'd like to finish the design. Anyway, I implemented bot ADHV4702 + my discrete driver.

hartytp, 2019-12-01:
Cool. Do you have a schematic preview? I won't have time to do a design review, but I'm just curious...

gkasprow, 2019-12-01:
@hartytp  Here are very initial, not finished schematics
[stabilizer_current_sense.PDF](https://github.com/sinara-hw/Stabilizer_Piezo_Driver/files/3908742/stabilizer_current_sense.PDF)

dtcallcock, 2020-01-05:
> I implemented bot ADHV4702 + my discrete driver.

What do you think your driver will offer over the ADHV4702?  

Will the two options be population options or will there be a switch or jumper?

dtcallcock, 2020-01-05:
Is it worth running anything to the GPIO header?  Perhaps just the status LEDs so that Stabilizer can throw an error if the amp shuts down?

gkasprow, 2020-01-05:
I implemented this discrete driver mainly for fun. We will probably leave only ADHV chip.
My driver can easily go +/- 60V into the 1nF load at 1MHz and can be easily scaled by changing the MOSFETs. Driver can be selected by the jumpers.