sinara-hw/Stabilizer_Piezo_Driver #3: Form Factor

dtcallcock, 2019-09-07:
Whilst a stabilizer AFE would be the most convenient in some circumstances it would preclude use with [Pounder](https://github.com/sinara-hw/Pounder/issues) or [Driver](https://github.com/sinara-hw/meta/issues/37) if they end up being Stabilizer AFEs.

How about just making it a standalone board with 2x input/output on the FP and barrel connectors for power?

We could make the board-board connectors a population option to retain AFE capability.

hartytp, 2019-09-07:
How would you configure it?

hartytp, 2019-09-07:
Or, is this just a 2 channel, fixed-gain (DIP switches?) HV amp?

dtcallcock, 2019-09-07:
Yeah, this board would be super-simple.  If you want bells and whistles there will be Zapper at some point.

DIP switches to select between a few different ranges and gains would be useful if easy to implement.

hartytp, 2019-09-07:
What use cases are you thinking of for this, and where would the D/A be?

dtcallcock, 2019-09-07:
> What use cases are you thinking of for this

My idea is to have the simplest possible board that would allow Stabilizer to drive piezos.  The only reason not to make it specifically a Stabilizer AFE is that it would block you from using other AFEs.  

> , and where would the D/A be?

This would just be a completely analog voltage amplifier.

hartytp, 2019-09-07:
> This would just be a completely analog voltage amplifier.

I meant what would drive it, but you answered that anyway.

> My idea is to have the simplest possible board that would allow Stabilizer to drive piezos. The only reason not to make it specifically a Stabilizer AFE is that it would block you from using other AFEs.

hmmm. Are you saying that you would use the same Stabilizer card to drive Zapper that you also use with another AFE?

dtcallcock, 2019-09-07:
> hmmm. Are you saying that you would use the same Stabilizer card to drive Zapper that you also use with another AFE?

Yes.  For example, for a doubling cavity lock (or locking a fiber laser to an iodine reference cell) you would have a PDH AFE on Stabilizer and then this piezo driver board between Stabilizer and the cavity piezo.

> drive Zapper

This isn't [Zapper](https://github.com/sinara-hw/zapper/issues/1).  Right now Zapper looks like it will be [way overspecced](https://github.com/sinara-hw/zapper/issues/1#issuecomment-497932589) for this common use case.  Obviously the Zapper spec is not set in stone yet though.

Perhaps the name of this repository should be changed to 'Stabilizer piezo driver' or something.

hartytp, 2019-09-07:
> Yes. For example, for a doubling cavity lock (or locking a fiber laser to an iodine reference cell) you would have a PDH AFE on Stabilizer and then this piezo driver board between Stabilizer and the cavity piezo.

Okay, makes sense. So this allows you to cover that case with only a single stabilizer. Fair enough.

gkasprow, 2019-09-20:
Let's draft specification:
- 3U form factor with panel, board size identical to current sense mezzanine.
- 2 HV channels
- ADHV4702 + my [discrete component driver](https://github.com/sinara-hw/Stabilizer_Piezo_Driver/issues/1#issuecomment-505290412) + DIP switch for configurable gain. This would let us test things for Zapper.
- 2x SMA input (maybe MCX?)
- 2x SMA output
- DC jack + 4 pin Molex supply connector on the back
- optional pin header for direct attach to the Stabilizer as a mezzanine
- 2x DC/DC converter [ R12-100B](https://www.digikey.com/en/datasheets/recom-power/recom-power-rxx-b) which can be adjusted +50...+135. We can make it adjustable by DIPswitches and simple protection circuit which makes sure we never exceed ADHV4702 safe operation range of 220V. Do we need to support asymetrical operation?

gkasprow, 2019-09-20:
do we plan to add some current limit protection for the ADHV4702 ?
It has internal temperature sensor. It could be used to disable power and turn on big, red LED on the front panel. The chip has also temperature protection that needs to be enabled.

dtcallcock, 2019-09-20:
> 2x SMA input (maybe MCX?)

Probably makes sense to use SMA so that a short SMA jumper can be used.

> 2x SMA output

Often slightly less common connectors are used on HV supplies to prevent accidentaly connecting the wrong thing and destroying it. I have seen SMB, SMC, LEMO OO and LEMO 0B used.

> DC jack + 4 pin Molex supply connector on the back

The DC jack would be on the front?  If so, would you be able to use a short barrel-barrel cable to power it from a Stabilizer running on POE?

> DC/DC converter R12-100B which can be adjusted +50...+135

150 and 200V piezos are also out there.  I think +135V is close enough to 150V and 200V piezos are uncommon enough that this is acceptable.  However if there is a different DC/DC that will do 50-150V  or 50-200V that would be better.

Otherwise looks great.

dtcallcock, 2019-09-20:
> do we plan to add some current limit protection for the ADHV4702 ?
It has internal temperature sensor. It could be used to disable power and turn on big, red LED on the front panel.

That seems to be how the datasheet is recommending you implement short circuit protection.  Sounds reasonable to me.

gkasprow, 2019-09-20:
> 
>     2x SMA output
> 
> Often slightly less common connectors are used on HV supplies to prevent accidentaly connecting the wrong thing and destroying it. I have seen SMB, SMC, LEMO OO and LEMO 0B used.

LEMOs are expensive. They are woth their price, but still. SMB/SMC is fine

> 
>     DC jack + 4 pin Molex supply connector on the back
> 
> The DC jack would be on the front? If so, would you be able to use a short barrel-barrel cable to power it from a Stabilizer running on POE?

Yes, but there is also 12V on the mezzanine connector

> 
>     DC/DC converter R12-100B which can be adjusted +50...+135
> 
> 150 and 200V piezos are also out there. I think +135V is close enough to 150V and 200V piezos are uncommon enough that this is acceptable. However if there is a different DC/DC that will do 50-150V or 50-200V that would be better.

Thre is compatible R12-150B that goes +92...+200.
So, the question is if we should make it to support asymmetrical operation.
For example, the user can select let's say +150, -50V. Then shall we add some adjustable offset to the input or simply sacrify the DAC resolution ?
> 
> Otherwise looks great.

gkasprow, 2019-09-20:
The evaluation kit of [ADHV ](https://www.analog.com/media/en/technical-documentation/user-guides/EVAL-ADHV4702-1CPZ-UG-1444.pdf)chip has on-board DC/DC converter based on [LT8304 ](https://www.analog.com/media/en/technical-documentation/data-sheets/8304fa.pdf). We can copy the design and make it operational over wider voltage range than R12-xxxB modules.

gkasprow, 2019-09-20:
Such solution can generate symetrical outputs. But I'm not sure if it is worth the effort.  The Recom modules are not that expensive (38EUR/pc)

dtcallcock, 2019-09-23:
>So, the question is if we should make it to support asymmetrical operation.

I've never had a need for piezo drivers that goes negative.

hartytp, 2020-01-08:
Was a decision made on the form-factor for this? IIRC it was decided (and I think it's a good idea) to make this a stand-alone card with a FP power connector, rather than a stabilizer mezzanine. Is that correct/still the case?

If so, this project should have a name change really. Also, what's the differentiation between this and the Zapper proposal? Just that Zapper is bipolar? Or, should they be merged?

hartytp, 2020-01-08:
@dtcallcock out of curiosity, what are your plans for using this for OSAs? What will you do for the transimpedance front-end for the PD? Will you design a small PCB to go right by the cavity? If so, how do you plan to power it? Use an external supply or break out the Stabilizer supplies somehow? Just curious...

gkasprow, 2020-01-08:
This module can either work as a Stabilizer mezzanine or stand-alone 2-channel amplifier.
This design is a 2-channel amplifier while Zapper will be multi-channel EEM.

hartytp, 2020-01-08:
> This module can either work as a Stabilizer mezzanine or stand-alone 2-channel amplifier.

Okay, nice :)

> This design is a 2-channel amplifier while Zapper will be multi-channel EEM.

I see. How many channels will Zapper have?

hartytp, 2020-01-08:
Answering my own question: [8](https://github.com/sinara-hw/zapper/issues/1)

How realistic is that from a power/component density point of view.

Anyway, I guess that there are enough differences between the two designs that it doesn't make sense to try to unify them.

gkasprow, 2020-01-08:
I treat the Piezo Driver as a prototype of Zapper. We will test PSU, 2 amplifier conceptions, protections.
If we go for ADHV4702, we can have even 32 channels. Some limitation is the output connector which needs to work with HV. There are [industrial rectangular connectors](https://www.tme.eu/html/EN/hi-temp-connector-inserts-180-degc/ramka_13219_EN_pelny.html) that can handle such voltage.

dnadlinger, 2020-01-08:
Not wanting to derail this discussion here, but for open-loop applications, charge-based drive might also be interesting to improve linearity/reduce hysteresis. (One use in AMO labs would be e.g. for multi-axis positioners/mirror mounts, although predictable response matters less there than for microscopy or trying to make absolute changes to lens positions in an imaging system.)

dtcallcock, 2020-01-08:
> @dtcallcock out of curiosity, what are your plans for using this for OSAs? What will you do for the transimpedance front-end for the PD? Will you design a small PCB to go right by the cavity?

Yes.  Perhaps Nadlinger's board.  Perhaps just an eval board.  

> If so, how do you plan to power it? Use an external supply or break out the Stabilizer supplies somehow? 

There was talk of putting some kind of power connectors (eg. M8-3) on here (or Stabilizer/Pounder), but there doesn't seem to be the front panel space for it.  For now I'll probably go with a bench PSU.  I have been thinking it might be worth developing a Eurocard PSU for Sinara though.

> I guess that there are enough differences between the two designs

There was an idea that Zapper would have onboard DACs and and EEM interface too.  Obviously the spec is still up for grabs.

> If we go for ADHV4702

A range of >200V was envisaged but if the ADHV4702 works great we should revisit that requirement.

gkasprow, 2020-01-08:
Btw, the 3U PSU is under development. One can see some drawings [here](https://viewer.autodesk.com/id/dXJuOmFkc2sub2JqZWN0czpvcy5vYmplY3Q6YTM2MHZpZXdlci90NjM3MTMyMTY5NDI3NjA3MzM4X2E5YjQzMDEzLTg4ZWQtNGVhZC05ZTM4LWUzODFlNzcyYWUyNy5jb2xsYWJvcmF0aW9u?sheetId=Yjg5ZDIxNzUtNmExNS00YmMwLWJjOWUtNjFhMmU0ZjA2MzQz&designtype=collaboration)
It is cooperation with CERN, we develop a low-cost alternative to COTS solutions.
One is 230V, 400W
Another is 18-72V, 200 or 400W. They fully conform with CPCIS including ORing and current sharing. The AC version has an IEC connector on the front panel. It also supports power delivery via a backplane connector. There is an option to produce it without any active circuits, just 12V. 
We can make a version with just DC jacks on the front panel and Molex connectors on the back panel if there is enough interest.

dnadlinger, 2020-01-08:
> If so, how do you plan to power it? Use an external supply or break out the Stabilizer supplies somehow?

One option would be to settle on a common "sensor" connector for the ecosystem (e.g. photodiodes, external preamps, …). We could just use a 4-pin LEMO connector and two twisted pairs (+12/power ground, signal/signal ground) if we are careful with conducted EMI from the DC/DC converters we'd invariably need on the peripherals.

gkasprow, 2020-01-08:
About the standardized connectors, LEMOs are great but expensive. Would 4 pins be sufficient? What about M8? They are low cost, robust and support up to 8 pins.

gkasprow, 2020-01-08:
@dnadlinger Do you mean HV amplifier with the current output? Let's start a discussion about it in /meta

hartytp, 2020-01-08:
https://www.piezodrive.com/wp-content/uploads/2016/01/IntroToCharge.pdf

gkasprow, 2020-01-08:
We could try to implement it as a feature of the Piezo Driver, but we would need other connectors. Moreover, I think such a driver needs to be matched with piezo to maintain stability.

hartytp, 2020-01-08:
> There was an idea that Zapper would have onboard DACs and and EEM interface too. Obviously the spec is still up for grabs.

If the DACs were really the only difference I'd have thought it would make sense to unify the design. The cost of adding a couple of SPI DACs and an EEM connector to this design is basically nill.

However, it sounds like there are more differences than that, so the designs should remain separate. I don't feel strongly either way, just interested to understand the design goals.

hartytp, 2020-01-08:
> One option would be to settle on a common "sensor" connector for the ecosystem (e.g. photodiodes, external preamps, …). We could just use a 4-pin LEMO connector and two twisted pairs (+12/power [...]

:+1: