sinara-hw/Stabilizer_Piezo_Driver #4: Power supply

gkasprow, 2019-12-01:
I plan to make the design quite universal. What about using 2 power supplies, each delivering 50..135V or 92..200V depending on assembly variant.
Then the user can choose the voltage range, both positive and negative by toggling the two rotary switches. One of 4 values would be possible within tuning range of each power supply.
In this way, any combination of upper and lower rails would be possible
So one can have for example
- 0-200 (one DC/DC module only)
-50..+50
-100..+100
-200..0 (one DC/DC module only)
-75..+75, etc.
-50...+150
There is simple protection circuit that checks if the Vpos - Vmin <200V. If it's true then power is disabled and an alarm LED is lit.
The rail voltages are signalized using 8 LEDs on the front panel.
In addition, an offset can be added to the input amplifier to use the DAC range fully. 16-position hex rotary swich would do the job.
Maybe one DIP switch to choose the gain.
In this way, we can satisfy most use cases by just adding a few resistors and 3 switches.
The module can operate stand-alone supplied from a single 12V DC jack or 4 pins Molex connector.
It can be also supplied from Stabilizer where DAC outputs are routed to the inputs.

gkasprow, 2019-12-01:
Edit: to simplify the design, the switches can be placed on the panel. No LEDs are needed.
The only LEDs woult be PowerOK and Overvoltage alert
[image]

dtcallcock, 2020-01-05:
> 50..135V or 92..200V depending on assembly variant

How about using something that can do 0-200V to avoid having variants:

https://www.xppower.com/portals/0/pdfs/SF_C_Series.pdf
https://www.xppower.com/portals/0/pdfs/SF_CA_Series.pdf

These are ~$150 compared to the ~$50 of the Recom module but if there is only one (see below) and population options are avoided, then that seems reasonable.  They look like they are lower EMI/ripple too.

> both positive and negative

I'd be happy to drop the negative voltages.  This simplifies the design as the offset is fixed and the gain and output voltage can just be controlled by the same single dial.  The overvoltage detection circuit can also be omitted.

Does anyone have a use case for negative voltages?

jbqubit, 2020-01-06:
Piezo response becomes highly nonlinear < -20  V. There's no practical application I can see for a negative rail below -20 V. See Newport [tech note](https://www.newport.com/medias/sys_master/images/images/hf1/hc8/9046758588446/Tech-Note-Piezo-Theory-Physics-and-Design.pdf).

dtcallcock, 2020-01-06:
>There's no practical application

There are exotic piezo stacks like [tube scanners](https://www.piezodrive.com/actuators/piezoelectric-tube-scanners/) that require symmetric +/-V.  I've never seen one used in an AMO lab and given the number of commercial solutions out there I think the emphasis for this board should be simplicity rather than covering all possible uses.

>negative rail below -20 V

I note that some commercial piezo drivers go slightly (eg. 10-20V) negative, but most do not.  I don't think this is a feature worth increasing circuit complexity for (or necessarily even desirable).

jbqubit, 2020-01-06:
Suppose a piezo with 100 V max rating. I can think of two aspects in favor of -20V. 1) 20% increase in travel 2) 20% lower peak voltage for a given travel (lower stress may increase piezo lifetime). I've certainly encountered circumstances where that 20% increase in scan range made my life much easier. 

I don't feel strongly about it.

gkasprow, 2020-01-06:
The amplifier is not rail-to-rail type. It needs a certain negative supply to support operation from 0.

dtcallcock, 2020-01-07:
The ADHV4702 can get within 1.5V of the rail.  Can we run V_EE off the planned -5V rail?

gkasprow, 2020-01-07:
Yes, we can and we should.

JakubKajzer, 2020-08-03:
Hello
I have a few questions about the power supply:
1. Should we keep 50..135V/92..200V  high voltage supply variants or "using something that can do 0-200V to avoid having variants"?
2. Just to be sure, only negative voltage we need is -5V?