sinara-hw/Stabilizer_Piezo_Driver #5: schematic review

gkasprow, 2020-11-06:
@JakubKajzer 
- the amplifier gain depends on the source output impedance. did you take it into account?
- connecting outputs of two amplifiers is not a good idea.
- did you simulate the signal chain with offsets?
- if you need components that are missing in the libs, let me know
- why R21 is 47Ohm?
- it would be good to add some signalization of overtemperature. Are you sure that the connection between TMP and SD is a good idea?
- the idea behind the latch-up circuit was to protect the amplifier before too high a voltage, not temperature.
- what is the role of IC10?
- do you think the 5V rail has low noise and stability so can be injected to the OFFSET input?
- will OPA197 work with output close to rails under the load?

JakubKajzer, 2020-11-27:
>the amplifier gain depends on the source output impedance. did you take it into account?

Yes, I simulated an amplifier with various capacitance at output. At lower gain, this problem is more visible, but due to significant series resistance at the output, the max gain peak is about 3dB at about 300kHz

>connecting outputs of two amplifiers is not a good idea.

Sorry for that, my mistake. It is a good idea, to make an optomos switch at outputs instead of switching inputs?

>did you simulate the signal chain with offsets?

Yes, offset is amplified with the same gain as an input signal.
In a discrete driver, the gain is 100x, so the offset can easily saturate the driver. Should I lower the gain?

>why R21 is 47Ohm?

I don't know what capacitances will be driven, so I decided to choose 47Ohm. This value should be less when driving bigger capacitances and should be higher when driving smaller capacitances (according to a datasheet of ADHV4702-1).

>it would be good to add some signalization of overtemperature. Are you sure that the connection between TMP and SD is a good idea?

A red led on the front panel will be enough? Connecting TMP and SD together, according to the datasheet, provides overtemperature protection.

>the idea behind the latch-up circuit was to protect the amplifier before too high a voltage, not temperature.

I added overtemperature protection because there wasn't any for the discrete driver. Should I add overvoltage protection?

>what is the role of IC10?

I added IC10 to negate LM75 outputs, so LM75 could work without changing its internal registers.

>do you think the 5V rail has low noise and stability so can be injected to the OFFSET input

Now I see that I should add a pi filter to offsets supply.

>will OPA197 work with output close to rails under the load?

OPA197 can provide about 2mA at the output with a 0.1V swing from the power supply.

gkasprow, 2020-12-16:
> Yes, I simulated an amplifier with various capacitance at output. At lower gain, this problem is more visible, but due to significant series resistance at the output, the max gain peak is about 3dB [...]

I meant SOURCE impedance, not the impedance of the amplifier

> 
> >connecting outputs of two amplifiers is not a good idea.
> 
> Sorry for that, my mistake. It is a good idea, to make an optomos switch at outputs instead of switching inputs?

you need to switch both.
Just use DIP switches to choose which one is connected to input and output. Make sure both cannot be selected

> 
> >did you simulate the signal chain with offsets?
> 
> Yes, offset is amplified with the same gain as an input signal.
> In a discrete driver, the gain is 100x, so the offset can easily saturate the driver. Should I lower the gain?

yes, use full DAC range

> 
> >why R21 is 47Ohm?
> 
> I don't know what capacitances will be driven, so I decided to choose 47Ohm. This value should be less when driving bigger capacitances and should be higher when driving smaller capacitances (according to a datasheet of ADHV4702-1).

OK, make it big, 0805 or so

> 
> >it would be good to add some signalization of overtemperature. Are you sure that the connection between TMP and SD is a good idea?
> 
> A red led on the front panel will be enough? Connecting TMP and SD together, according to the datasheet, provides overtemperature protection.

OK, use red LED.

> 
> >the idea behind the latch-up circuit was to protect the amplifier before too high a voltage, not temperature.
> 
> I added overtemperature protection because there wasn't any for the discrete driver. Should I add overvoltage protection?

If there is a risk that the user can setup dangerous voltage for driver IC, add such protection.

> 
> 
> >what is the role of IC10?
> 
> I added IC10 to negate LM75 outputs, so LM75 could work without changing its internal registers.

since we don't need such circuit for temperature protection, skip it.

> 
> >do you think the 5V rail has low noise and stability so can be injected to the OFFSET input
> 
> Now I see that I should add a pi filter to offsets supply.

please add low noise LDO

> 
> >will OPA197 work with output close to rails under the load?
> 
> OPA197 can provide about 2mA at the output with a 0.1V swing from the power supply.

OK
> 
> 
>

JakubKajzer, 2021-01-01:
> I meant SOURCE impedance, not the impedance of the amplifier

I checked this at simulation and due to low input impedance, at input must be placed a voltage follower like in discrete amplifier.

gkasprow, 2021-01-13:
sure, you can increase the supply to +/-12V. It is already available from Stabilizer.
Yes, change the MOSFET, but use one with similar specs as the low-side one.

JakubKajzer, 2021-01-20:
I've uploaded updated schematics.

dtcallcock, 2021-03-05:
@JakubKajzer Can you post a pdf please?

JakubKajzer, 2021-03-06:
@dtcallcock 
Of course
[schematic.PDF](https://github.com/sinara-hw/Stabilizer_Piezo_Driver/files/6096184/schematic.PDF)